FrSkyRC/RF_Feedback_Group #109: Update TW GR8 receiver from 1.07 to 1.09 failed

Florentus, 2025-01-30:
I use Ethos 1.5.9 on X20PRO AW
Before upadating the receiver TW GR8 , version was 1.07. Registration and binding went very well. I updated the firmware to 1.09 using the Sport cable and the flashing is ok. I did not use OTA. It was after that the problems started: the receiver not turn on, no LEDs lit , even the red LED. (receiver disconnected from the Sport and external battery connected to it)
Using the same procedure I downgraded the receiver to version 1.07 and it started working again.
I did not try to flash it to 1.09 again

Florentus, 2025-01-30:
Thanks for your help. 
My RF module is in 1.1.13 because 1.1.14 had issues and was removed from the download list in Ethos suite.
I don't think the problem comes from the RF module because I updated with a cable via the Sport of the transmitter. After update **no LED of the receiver lights up** as if it was powered off. I suppose the problem was caused by a bad flashing as it has already been reported using the OTA update, that's why I used a cable and the Sport
I will redo the update to check if the problem is systematic or random.

Florentus, 2025-01-30:
I just updated the receiver again but using OTA. This time everything went well, the receiver works fine with version 1.09. (I did it on 2 receivers)
What is worrying is that I had the same update issues using the Sport cable as users who had damaged receivers with the OTA update.
It seems  the receiver is writing corrupted data if there is a data transfer error. ??

SmartTommy, 2025-02-05:
> I just updated the receiver again but using OTA. This time everything went well, the receiver works fine with version 1.09. (I did it on 2 receivers) What is worrying is that I had the same update [...]

hi, don't worry. please flash your Rx  with sport cable again. (Sport cable flash will not damage receivers)