FrSkyRC/RF_Feedback_Group #110: TD SR18 Channels Not stabilized

4qr1hu9n7, 2025-02-04:
Hello.
Channels 5 and 6 are not stabilized.

[image]

bsongis-frsky, 2025-02-04:
I forward this issue to the RF team. I think it is a better place as it is not related to Ethos

doppleganger53, 2025-02-04:
> Thanks to everyone for your input. I'll just pass on this feature. I may come back to it some other time.
> 
> thanks again.

The stabs do work, I have 2 models that use stabilization on 10 channels. You either have misconfigured something or have a defective receiver. Either can be addressed, but you're going to have to provide more details about how all of your channels are configured, how the Rx output pins are configured, etc. 

If you do want to tackle it again, I'd suggest posting on the facebook group and including a video demonstrating the issue and your setup if possible, that group is extremely helpful.

closed 2025-03-06 by SmartTommy