FrSkyRC/RF_Feedback_Group #111: RSSI 2.4G still „received“ and displayed after TW R8 is switched off

glipski50, 2025-02-10:
X20R, Ethos 1.6.1 LBT, ISRM v3.0.1
TW R8 v3.0.1

After turnig off the receiver TW R8 and waiting **5 min** the telemetry page still shows a RSSI 2.4G signal beeing received although LoRa 2.4G is displayed red i.e. no signal. RSSI 2.4G is also displayed in the screen head line with all bars. In the telemetrie display the number for "EMPF." (REVEIVER) is constantly counting up 0-1-2-0-1-2 in ca. 20 ms intervall.

[image]

KAID-YANG, 2025-02-12:
Hello, do you remember your operation steps? Did you perform the "reset bind" operation?

glipski50, 2025-02-12:
I am not 100% sure what you mean with „reset bind“. 

I had my X20R on Ethos 1.6.1 pre-Release and the ISRM and TW R8 on 3.0.1 RC2. I then updated with the Ethos Suite to 1.6.1 final and to 3.0.1 final. I think I then had to do a bind since the TW R8 was not recognized after the update and on the screen I got the options ???? - Bind - Delete Binding.

KAID-YANG, 2025-02-12:
[image]

glipski50, 2025-02-12:
ok. understood. Receiver was bound normally and I then simply disconnected power directly.

glipski50, 2025-02-18:
During testing my GPS ADV, just RE R8 + GPS ADV, i had that mentioned issue several times again. After switching off the receiver power I could see the RSSI 2.4 bars as active. Even after 30 minutes.One time today it even diplayed both RSSI 2.4G and RSSI LoRa both with all bars as active.

[image]

When I finally wanted to swich off the receiver as well I received the message that the receiver is still switched on.

KAID-YANG, 2025-02-19:
I think we have found the issue, but we have never encountered a situation where FSK and LoRa continue to work after the receiver power is disconnected. Can you reproduce this phenomenon?

glipski50, 2025-02-19:
It seems to be an intermittend topic and I cannot reproduce it easily. I‘m going to set up more models the nex days and I will watch intensivly the steps to see when the issue comes up again. Maybe others will do the same.

bsongis-frsky, 2025-02-24:
@glipski50 No need to open new issues about the same problem!

SmartTommy, 2025-02-25:
The TD-ISRMpro & TD-ISRM & TD-ISRMRs  3.0.2  released,  fixed this

strgaltdel, 2025-02-26:
> The TD-ISRMpro & TD-ISRM & TD-ISRMRs 3.0.2 released, fixed this

Hi SmartTommy,
i can't find the TD-ISRMpro 3.0.2:

wether on git,
https://github.com/FrSkyRC/RF_Feedback_Group/releases

nor on FrSky's Site
https://www.frsky-rc.com/tandem-x20-pro-aw

where can i download the actual 3.02 public-release ?

Thanks

SmartTommy, 2025-02-27:
> Hi SmartTommy, i can't find the TD-ISRMpro 3.0.2:
>
> wether on git, https://github.com/FrSkyRC/RF_Feedback_Group/releases
>
> nor on FrSky's Site https://www.frsky-rc.com/tandem-x20-pro-aw
>
> where can i [...]

[image]

LiangAndrea, 2025-02-27:
> Hi,
> 
> thanks for your reply, the mentioned TD-ISRMpro 3.0.2 Firmware targets the internal Rf module of X20pro Transmitters, which i do use and wanted to update i couldn't find the corresponding files
> 
> unfortunately the download shown is for for TD R18 Receivers, so i do assume there is no update available for x20Pro RF modules by know, right ?

[image]

sigurdma, 2025-02-27:
This is RC1, best to wait for the official release?

glipski50, 2025-03-02:
Will there be a ISRM RF 3.0.2 RCx for the X14 transmitter available? At the moment the Ethos Suite only shows version 3.0.1.

[image]

Dsrenger, 2025-03-03:
> The TD-ISRMpro & **TD-ISRM** & TD-ISRMRs 3.0.2 released, fixed this

At the moment there is no 3.0.2 (RCx) wir the TD-ISRM module. Just for the Pro und RS module.

Is that correct?

SmartTommy, 2025-03-04:
> > The TD-ISRMpro & **TD-ISRM** & TD-ISRMRs 3.0.2 released, fixed this
> 
> At the moment there is no 3.0.2 (RCx) wir the TD-ISRM module. Just for the Pro und RS module.
> 
> Is that correct?

yes, it will be released next week.

enzo-molinari, 2025-03-07:
Will there also be a 3.0.2 release for the normal Pro without AW?

Dsrenger, 2025-03-07:
its there. the 3.0.2 for TD-ISRM R und TD-ISRMpro HF modules got released today

[image]

sigurdma, 2025-03-07:
It solved active after power off. I still get no voltage in telemetry. That may be the rx TD SR12 on 3.01. (Only TD R18 has 3.02)

enzo-molinari, 2025-03-07:
After the firmware update 3.0.2 for the X20Pro, the suite shows that the module version is greater than the latest release.

[image]

SmartTommy, 2025-03-08:
> Will there also be a 3.0.2 release for the normal Pro without AW?

both Pro and ProAW use the same TD-ISRMPro Tx

SmartTommy, 2025-03-08:
> It solved active after power off. I still get no voltage in telemetry. That may be the rx TD SR12 on 3.01. (Only TD R18 has 3.02)

did you mean your radio no show the RxBat telemetry of the TDSR12? it shows normal here.

could you try to discovery the telemetry again? 

[image]

strgaltdel, 2025-03-08:
> After the firmware update 3.0.2 for the X20Pro, the suite shows that the module version is greater than the latest release.

pretty sure that's why 3.02 ist still "release candidate", not "full release",
have a look at the filename: TD-ISRMPro_3.0.2_RC1.fsk

Perhaps it should be labeled as “RC” in the Suite Download Manager so that users can distinguish between the statuses.

Dsrenger, 2025-03-08:
I am pretty sure it is supoessed to be the full released.
IT was marked as RC1 in the suite last week. The RC mark in the suite disappeard, when the release note on the main homepage appeard

RCs dont get a note on the main homepage.
Guess they missed to rename the fsk file ;)

[image]

strgaltdel, 2025-03-08:
Dennnis, you could be right:
both files (from last week and now) seems to be bit identical (checked with fc.exe).

maybe no more issues were reported and they decided to "give the go" for rc1 as full release.

sigurdma, 2025-03-09:
> > It solved active after power off. I still get no voltage in telemetry. That may be the rx TD SR12 on 3.01. (Only TD R18 has 3.02)
> 
> did you mean your radio no show the RxBat telemetry of the TDSR12? it shows normal here.
> 
> could you try to discovery the telemetry again?
> 
> [image]

Little off topic here, but I have done discover telemetry first and then reset RX, and then re register and bind again. No help.. All other telemetry is there, just missing rx volt. I f i see in the black box on rx it measures 7.x V as it should.

glipski50, 2025-03-09:
I‘m not sure if this is still the same issue as I originally posted.

X20R Ethos 1.6.2 TD-ISRM Pro 3.0.2
TW R10 3.0.1

I‘ve done a range check. After switching off the receiver the range check still shows 2.4G VFR 100% and LoRa VFR 91%, even after 2 hours. When I switch off the transmitte and restart both VFR are 0% as expected. 

[image]

Dsrenger, 2025-03-12:
> The TD-ISRMpro & TD-ISRM & TD-ISRMRs 3.0.2 released, fixed this

I guess the Twin Lite Pro module will get a new firmware as well?

If so, I will wait with an update until then.
3.0.2 seems to have most of the bugs ironed out.

Burgerman-X10express, 2025-03-14:
Same on 900 on flex.  In this case as a added redundancy rx on Access. 
So once powered up it sits at 10mW output and high RSSI. As it should. 
Disconnect power from Rx and it still senses a high RSSI and so keeps transmitting at low power. 

So if that were to happen due to interference or flying behind some RF blocking trees or building or whatever it does not jump up to the 1W level as it should. So you dont get the advantage of the 1W power to reconnect the aircraft/drone. It just sits uselessely at 10mW...