FrSkyRC/RF_Feedback_Group #114: TDR18 RX Current not working

Maddmaxx61, 2025-02-12:
After updating TX to 1.6.0 and TDR18 RX to 3.01 the RX current doesn’t work. I deleted all sensors and re learned them but the RX current will never show anything other than zero. I tried changing from amps to miliamps and also showing more decimal places but it still shows zero. I also tried flashing the software to the RX again but still have the same result.
Is this a known bug in 1.6.0?

Today I updated the X20S to 1.6.1 in hopes that this version fixed the RX current not reading but this didn’t fix the problem. My guess is that the problem is in the RX 3.0.1 software?

bsongis-frsky, 2025-02-12:
Not an Ethos issue, rather RF I think

pharaohrs, 2025-02-12:
I know at least 2 more users that had this problem,  than they downgraded the system (radio+rx) and it's solved . I haven't access to my computer now to do this but I'll try to downgrade ... Current is a important thing to me in the tdr18 applications.

SmartTommy, 2025-02-13:
3.0.2 fixed this issue 

[TDR18_3.0.2_RC1.zip](https://github.com/user-attachments/files/18775498/TDR18_3.0.2_RC1.zip)

pharaohrs, 2025-02-13:
> 3.0.2 fixed this issue
> 
> [TDR18_3.0.2_RC1.zip](https://github.com/user-attachments/files/18775498/TDR18_3.0.2_RC1.zip)

Yep , tested 3.0.2 here , it's solved . Current is fully functional now . Thanksss man !

Maddmaxx61, 2025-02-13:
Works Great. Thanks for the help.

SmartTommy, 2025-03-06:
3.0.2 fixed