FrSkyRC/RF_Feedback_Group #115: I can't update to the latest 1.6.0 firmware version

Igi82, 2025-02-15:
I have a Turnigy X14 transmitter (firmware version 1.5.15 EU) and I can't update to the latest 1.6.0 firmware version. It says Missing remote version and does not display a version to choose from. How to solve this ?

[image]

KAID-YANG, 2025-02-17:
You can report ETHOS issues here: [https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues](https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues).  

Try manually downloading the corresponding firmware version from GitHub, enter **bootloader mode**, and copy **firmware.bin** to the root directory of the **RADIO** folder on your transmitter. Then, unplug the USB cable, and the firmware will automatically flash. Once the progress bar is complete, reconnect to **ETHOS Suite**.  

If the issue persists, try using a different USB data cable.

Nicholas-Luoyi, 2025-02-18:
This seems to be a network problem

Igi82, 2025-02-18:
Ok I downloaded it manually and it worked. Except that it no longer shows Turnigy but Frsky.

Nicholas-Luoyi, 2025-02-19:
Sorry. After the check, we found that the firmware for your radio is missing in the Release 1.6.1. Which should be T14-XXX.zip. We will fix it in 1.6.2 (not nightly). Please manually flash it after the 1.6.2 is released. After that Suite would work correctly.