FrSkyRC/RF_Feedback_Group #118: RF FW 3.0.2 RC1: Problem with sensors Temp1 / Temp2

strgaltdel, 2025-03-04:
Hi,

I have been using openX-GPS for many years.
The sensors "Tmp1 / Tmp2" with AppID 0x0400 / 0x0410 transmits the number of satellites and HDOP.

Under RF 3.02RC1 on an X20pro, this is causing issues:

Tmp1 does not work correctly, and Tmp2 does not work at all.

My X18 is still running on an older firmware. The GPS on an Access Archer GR8 detects all sensors, including Tmp1 & Tmp2.
According to the display, data from these sensors is updated several times per second.

However, the X20pro is experiencing problems under 3.02 RC1 with the same receiver:

After deleting the sensors, Tmp2 (0x0410) is not recognized at all.
When GPS is receiving: Tmp1 is detected, but the refresh (poll?) is slow. After a few minutes (around 3 to 5), no more data updates occur.
The problem is already present with the aforementioned Archer GR8.
It also occurs with Twin GR6 / Tandem R18, both running FR 3.02 RC1.

regards
udo

SmartTommy, 2025-03-05:
we don't have a opernX-GPS. could you help us do some test to find the problem?  flash your ethos,tx and  rx back the the version you used before. then update your Tx to 3.0.2, and connect to your Rx, try to find the GPS sensor. if works normal, update your Rx to 3.0.2, try to find GPS sensor. if still works, update your ethos.

strgaltdel, 2025-03-05:
Hi SmartTommy,

I will try it this evening.
By the way:
I don’t believe it’s related to the RX firmware because the Archer GR8 firmware hasn’t changed – I think it’s running on some 2.1.x version – and there are no issues when using it with my X18 (Ethos 1.5.19 and the corresponding RF firmware).

However, as soon as I bind this receiver setup to the X20pro, I lose Temp2 and experience issues with Temp1.

I will downgrade step by step.
There is a similar F.Bus issue in the Ethos group:
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/5202

Thorsten is from Germany, I know him from Andreas' forum.
He is a contributor to the oxs_on_RP2040 project.

I received the following information from him:
It seems that he was able to detect both sensors on 3.0.1 using S.Port, but he ran into scaling issues (by a factor of 100 or so), so he had to manually adjust the factor.

We are in the European time zone, so it will take a few more hours until I can perform further tests.

Thanks for your support!
Udo

strgaltdel, 2025-03-05:
**I was able to solve the problem,**  
it is not related to the RF firmware.  

Torsten / Satcomix had the crucial hint:  
The new Ethos revisions have set the allowed limit for temperature sensors set to a maximum of 110 degrees.  

That explains everything:  

- **Temp2**: As soon as a GPS gets its first locks, the HDOP is still quite high, above 110, which is why there is no sensor data on Temp2.  
- **Temp1**: With a lock, oXs sends an offset of 100 to indicate that reception is valid.  

At the beginning, with 6–7 satellites, the value is below 110.  
At some point, more than 10 satellites are received, and the sensor stops working.  

This applies to sensors that were automatically recognized (years ago).  
If they are deleted and set up as DIY sensors (which actually makes sense for "DIY"), the range can be increased,  

and voilà, everything works as it always should have!  

**Thanks & best regards,  
Udo**