FrSkyRC/RF_Feedback_Group #119: Independent switching of the stabilizer groups

WhiskeyGolf, 2025-03-05:
A small request for the next updates.
It would be very important to be able to switch the 2nd stabilizer group independently of the first group.
I absolutely need it in a jet where the steering servo continues to react in the stabilized state even when the landing gear is retracted. This causes high current peaks on the servo because it blocks.
One idea would be to choose whether to use channel 15 to control the 2nd group instead of the knifeedge/hover mode.
Many thanks in advance

doppleganger53, 2025-03-05:
Another very beneficial use case for this is for jets with vectored thrust. Its useful to be able to disable thrust vectoring but maintain stabilization on other surfaces, during takeoff and landing for example. Independent master gain per stab section would be very useful as well for tuning vectored thrust, steering, and other uses of separate stab groups . Would very much appreciate this feature.

SmartTommy, 2025-03-06:
we have add this and some new function in the new vesion RB35S firmware.  Once the testing is completed, we will release it. Then these new features will also be added to some receivers