FrSkyRC/RF_Feedback_Group #120: Wrong firmware file version

Alex00101, 2025-03-10:
Connected my X20 Aw Pro to the Suite and got notified that the ISRM module (currently on 3.0.1) is out of date and can be flashed to 3.0.2.
The actual frsk file is actually labelled TD-ISRMPro_3.0.2_**RC1**

Could you please either confirm this is production release and rename the file accordingly or label the release as RC1? 

Very confusing to those who don't want to load release candidate versions

ETHOSWarrier, 2025-03-11:
Agreed,my x20R rf firmware is the same...says 3.0.2 in the title(no RCx etc) but the actual file is also named RC1 like yours...very confusing...EW

KAID-YANG, 2025-03-17:
Hello, this is a stable version that has been tested. We will improve the firmware naming issue in future updates.

Burgerman-X10express, 2025-03-18:
Does this mean we should update to 3.0.2 RC1?
Still confused!

KAID-YANG, 2025-03-19:
This is just a naming error on our part. Please refer to the firmware on Suite and the official website as the correct version.