FrSkyRC/RF_Feedback_Group #121: TW SR12 rudder oscillate on the ground.

WhiskeyGolf, 2025-03-10:
The rudder on my GFK turbine jet starts to oscillate when stationary. This happens in the lower speed range of the turbine when the acro mode is switched on. 
The turbine has no noticeable vibrations (xicoy x45) and the receiver (TW SR12) is attached using the adhesive pad provided. 
My gain setting is at 50% and the rudder gain is at 60%. The phenomenon also occurs at the same time on the nose gear (2nd gyro group). 
Is there no LPF for the gyro to filter out high-frequency interference that is above the servo speed anyway?

noamkrief, 2025-03-10:
This is your solution.
https://www.youtube.com/watch?v=SK9qpGqBBmg

Forget the hush boxes etc. It's not about noise. Noise is vibration anyway. You want to kill the vibration and this tape does that job 100%.

WhiskeyGolf, 2025-03-12:
Thanks for the tip about the tape. 
However, the problem should be addressed more thoroughly to prevent unwanted oscillations during flight. 
So, the question is whether the STAB receivers have a gyro LPF, and where it kicks in. An LPF would also be good for protecting the servos and control surfaces from excessively rapid movements, since fixed-wing aircraft can't follow these high frequencies anyway. Incidentally, this also prevents the high current peaks in the elevator servos that occur when taxiing on rough surfaces.

KAID-YANG, 2025-03-18:
Hello, could you provide a related video of the issue and the model name of the device you are using so that we can analyze the problem?

WhiskeyGolf, 2025-03-18:
Unfortunately, I'm currently on tour and can't make the video until next month. The model is a Schreiner L39 made of composite material with an x45 turbine, Xact servos, and a TWSR12 receiver. Is that all the information you need?

WhiskeyGolf, 2025-04-03:
Is there any information on this problem yet? Where is the software-based gyro LPF? 
I also need to know if it's a hardware issue? I didn't have these issues with the S8R.

WhiskeyGolf, 2025-04-25:
Still no answer?

SmartTommy, 2025-04-27:
hi @WhiskeyGolf , please test the new firmware. i changed the cut off frequency of the LPF.

[TWSR12_3.0.3_Beta.zip](https://github.com/user-attachments/files/19928127/TWSR12_3.0.3_Beta.zip)

SmartTommy, 2025-04-29:
> Thanks for the firmware. Where is the LPF now? Is it possible to fly with the beta version? I'm currently touring a lot, but I'll test it as soon as possible.

The LPF always exists,i changed the cut off frequency to adapt to low-frequency vibrations.

WhiskeyGolf, 2025-04-29:
That's what I assumed, too. 
I just wanted to know the frequencies that were set before and now. Then I can compare them with an analyzer. 
Is the Beta flyable?

SmartTommy, 2025-04-30:
> I just wanted to know the frequencies that were set before and n

250HZ---->40HZ

you will get the release version next week.

SmartTommy, 2025-05-13:
does the new firmware sloved the rudder problem?

WhiskeyGolf, 2025-05-13:
As I said, I'm on tour for a long time right now. But I'll check it out at the beginning of next month. It'll be at the top of my list then.