FrSkyRC/RF_Feedback_Group #123: Ethos Suite 1.6.1

BJR981S, 2025-03-12:
I have found that with the MacOS version of Ethos Suit it wont release the lock on the Flash rive after exit,.

Also it has a tendency to trap if restarted in the state.

Sequoia 15.3.1

[image]

BJR981S, 2025-03-15:
Additional information on crash.

[image]

KAID-YANG, 2025-03-17:
The issue has been reported, and it is expected to be fixed in the next version.

BJR981S, 2025-03-17:
> The issue has been reported, and it is expected to be fixed in the next version.

Fantastic Thanks for the information.