FrSkyRC/RF_Feedback_Group #125: Archer r10+

beanie592, 2025-03-15:
Since update from 1.1.0 to 3.0 all both my archer r10+ will not work . Have been im chat in ethos git hub and was told to post here as they think its rf issue? Please see messages post3d from ehthos group below.
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/5237.

Basicly all my td rx work on latest update but my rx10+ will not work . Since ethos update. Im running x20 pro  . Using 1.6
2 . Isrm 3.0.2 ethos and archer r10+ on latest available firmware. As soon as i go to versiin 3.0 onwards both rx will not respond to tx but wirk as they should if flashed back to version 1.0.1
I cannot use these rx since the update. Any help would be apreciated.

strgaltdel, 2025-03-15:
> ....onwards both rx will not respond to tx but wirk as they should if flashed back to version 1.0.1

hi,
whats your understanding of ..will not respond?

during register rx? (did you try to re- register?)

during bind to template ?

they are bind to tx (indicated by green led), but dont deliver channel output signals ?

beanie592, 2025-03-15:
Hi the rx appears bricked and will not register and if trying to bind the tx does not reconise the rx at all. If put back to version 1 the rx registers and come up as rx10+ as normal but as soin as you add latest 3.0.1 both rx do not respond at all

beanie592, 2025-03-15:
At no point does the tx recognise the rx type when trying to register or bind the tx just waits for rx. Go back to version 1 and all works correctly

KAID-YANG, 2025-03-17:
Hello, did you upgrade the receiver using the S.Port cable or via OTA? We have tested the same version with the same receiver and did not encounter any issues with registration and binding. We would appreciate it if you could provide a video showing the receiver not working so that we can analyze the issue.

beanie592, 2025-03-17:
Hi i did the update via OTA TO 3.0.1 AND used cable back to version 1 as rx does not respond to OTA once version 3 is on. I will try and sort video Asap but working next 12 hours

beanie592, 2025-03-17:
Hi guys thanks for your help. Whilst making a vid for you it started working as it should. The only thing i have done prior to making a video was a full factory reset on my x20pro and  then re updated both rx to latest version via OTA AND ITS WORKING dont have any idea whats different as i had redone the tx and rx updates multiple times but the factory reset has cured the issue for me. Everything is now as it should be so i can only assume something was amiss with the origanal update i downloaded. Thanks for your help i will close this off.

closed 2025-03-17 by beanie592