FrSkyRC/RF_Feedback_Group #126: lost connection midflight.

chrols95, 2025-03-18:
antenna placement were 90 degres on themself. and outside fuselage.
here is the logfile.
hopefully this is a known bug or somethin. what to do?
everythin is up to date with rf module and on the reciever

Burgerman-X10express, 2025-03-18:
2.4g rf noise from millions of users on this licence free band, is now common. Every phone has wifi, BT, alarms, wifi routers, communication systems, door openers, farming control systems, even powerline control system now use it. Where I live and fly a solid 2.4ghz link happens 1 day out of 3. Fly anywhere near a farm or a bunch of houses and interference is a possibility. 
Thats why we now have frequency diversity. So a TD reciever, for e.g or a 900 tiny fingernail sized redundancy receiver. So if something takes out the 2.4g band like even a microwave oven or whatever then you still have a safe backup link.

T16-Pilot, 2025-03-18:
@Burgerman-X10express

I am flying at the same field as "chrols95"

Our field is in the middle of no where, fare from people, houses and so on. We don´t even have mains power. We are charging our battery packs with solarpanels. We are at 200+ meters above sea level in the "mountains" in southern Norway. Nothing but rock and forest around our airfield.

I also fly Frsky Ethos radios at the same field with out 900mhz for the most part.

I was at the field both times when my friend lost signal. I know this happened.

Burgerman-X10express, 2025-03-19:
Non of that exempts you from 2.4ghz RF interference. It can be bursts of very directional signals, or even caused by sattelites, or someone with a dish and a booster and those are common on aliexpress, 30 different brands. If 2 aircraft suffered this at the same time then realistically that must be what it is. Interference from some electrical noise or some 2.4 source.

robthomson, 2025-03-19:
> > You mentioned you pulled the rf module to another radio.
> > 
> > Exactly what rf module is this?
> 
> same rf module as the other radio has. was from an x20 to an x20s, so the rfmodule is exactly the same.

I am trying to understand something.

When you say you changed the rf module, do you mean you changed the software on it?

It are you using an external module in the back of the radio?

Or did you physically move the rf board from one radio to the other?

Joshadam0057, 2025-03-19:
This is a known issue ..1.6.x is trash, tgeres 3 other threads here with signal/range loss and it seems they dont care. I say that because tgey havent taken any action to resolve it

chrols95, 2025-03-21:
> I am trying to understand something.
>
> When you say you changed the rf module, do you mean you changed the software on it?
>
> It are you using an external module in the back of the radio?
>
> Or did you [...]

i had a spare radio (x20s) and just put that rf module inn the radio, the RF and VFR is lower (atleast i think it is) than usual, i dont know if its because of the new update..

Burgerman-X10express, 2025-03-21:
_i had a spare radio (x20s) and just put that rf module inn the radio, the RF and VFR is lower (atleast i think it is) than usual, i dont know if its because of the new update.._

He means the rf deck, the rf output board inside the TX, rather than an actual module I think?  Also I notice he is using a rf blocking carbon fusalage. So much more likely to see interference or shorter range.

sigurdma, 2025-03-22:
T16-Pilot, could try a rang test?

Burgerman-X10express, 2025-03-22:
I would put the Rx in a quad and fly it away and see how far it gets before RTH kicks in.
And would do it in several directions. 

On 2.4 fsk I get almost a mile in one direction and about half that in the opposite direction from my house.
On TW receivers the FSK part is the same. But the LoRa part just seems to go on until battery is gone...
Same with TD receivers.

If I fly a sub 250g quad out over the town with lots of wifi, phones, etc I get less than 1/4 mile on FSK.

**_His glider has a carbon fuse, but the antennas are on the outside of the fuse._**

If its the whisker type (no balun) antennas then 2 things happen.
1. Half the anttenna radiation cones from the coax. Right back to the reciever. It forma the other half on the antenna. And because its not "straight" away from the whisker it distorts the shape of the reception/transmission hugely. On a carbon receiver its also trapped inside!
2. The carbon itself detunes the SWR and the shape of the radiation pattern horribly. As well as blocking it over half of its pattern.

So making the results of the quadcopter tests above meaningless and much less reliable in various angles and distances.
Just a thought.  I measured all this with a Vector analyser.  

Antennas with a balun, hanging down away from the fuse will perform way way better!  So will a TW rx.

remcat07, 2025-03-22:
@Burgerman-X10express, thank you for your tests.
It would also be instresting to measure the impact of the orientation of the transmitter ( with internal antenna). I noticed that on the slope when my glider was under my feet ( yes it happens 😉) I got some problems which disappeared with external antenna.

Burgerman-X10express, 2025-03-22:
Well it wouldnt make much difference on my TX as theres 2 at 90 degrees. For that reason!
When using an external one on my TX its either or both LoRa 900 or LoRa 2.4. No external on the FSK.

Heres another thing... I have a twin lite pro module. That can be set for 500mW ACCESS or TW instead of the internal ones 100mW which doubles the feild strength at every point in space so 3 to 4db up everywhere. Or if you prefer more than double the range.