FrSkyRC/RF_Feedback_Group #128: Archer r10+

beanie592, 2025-03-18:
I am runing x20pro on ehos 1.6.1 and module 1.45 i have two new r10+ and flashed them from 1.10.1  to 3.0.1 and both no longer respond to the tx. I flashed them back to olf firmware and they work fine. I have uodated all my td rx and they are all fine. I reflashed moduel amd tx and redid r10+ but both the same they will not work with the latest firmware. I emalled eu support but have had no responce. Obviously i want these rx on latest firmware but can i use the latest ehos module and tx firmaware whilst still having r10+ running 1.10.1 or is there an issue . Im in Uk and have no issues with any other rx justvthe two archer r10+ maybale a hardware issue not matching latest firmware

KAID-YANG, 2025-03-19:
Is this the same issue as #125? Or is the Archer Plus R10+ unable to connect to the X20 PRO again?

beanie592, 2025-03-19:
Yes i closed it now rectified

closed 2025-04-29 by SmartTommy