FrSkyRC/RF_Feedback_Group #131: Intermittent zeroes on RPM readings with Ethos 1.6.1 and Advanced Engine Suite

3Dholic, 2025-03-19:
I equipped my DLE222 with an AES with 4 temp sensors and 2 RPM inputs...

Today, started the engine first time and got readings on 6 sensors above... However, RPM readings were showing intermittent zeroes...

Any idea why ?

Notes:
AES is updated to the latest firmware
TDR18 rx is updated to the latest firmware
Running Xact BLS 6505 x 9 servos on Fbus

3Dholic, 2025-03-19:
@bsongis-frsky, as you state that this is not an ETHOS issue, how can we justify the cause of problem ?

Is it AES or Ignition system ?

xet303, 2025-03-19:
> > > >
> 
> Checked again the other videos that I recorded... It doesnt go to zero only if it is below 1000... It occurs also ar higher speeds as well (ie 1400rpm)

Your YouTube video doesn't prove that the phenomenon occurs above 1000. A frequency meter will always have a given reaction time. Moving your Th stick this way doesn't tell us if at any given moment you're not going below 1000. Repeat your test at stable RPMs for a few seconds, and you'll see that the problem occurs below 1000.
Hopefully, I was helpful and have nothing more to add personally.

xet303, 2025-03-20:
Thank you! You're efficient!
I tried the 1.7-RC this morning with my coffee. The phenomenon below 1000 RPM is no longer there. However, now the RPM display no longer returns to zero, preserving the display of its last count. Only an AES power-up resets it to zero. I also noticed that a power-off now clears the remaining % of the tank as well as the remaining tank capacity. This wasn't the case before.
I posted a temporary video here in a panic. You should be able to view it if the entire planet doesn't pump it all at once.
[]https://lesdesjardins.org/2025-03-20-07-58-31-597r-1.7rc1%20%20%281%29.mp4(url)

3Dholic, 2025-03-20:
> Thank you! You're efficient! I tried the 1.7-RC this morning with my coffee. The phenomenon below 1000 RPM is no longer there. However, now the RPM display no longer returns to zero, preserving the display of its last count. Only an AES power-up resets it to zero. I also noticed that a power-off now clears the remaining % of the tank as well as the remaining tank capacity. This wasn't the case before. I posted a temporary video here in a panic. You should be able to view it if the entire planet doesn't pump it all at once. https://lesdesjardins.org/2025-03-20-07-58-31-597r-1.7rc1%20%20%281%29.mp4 https://lesdesjardins.org/2025-03-20-07-58-31-597r-1.7rc1%20%20(1).mp4.zip 126Mo
> 
> Nice progress,Thanks !

Watching your video, I saw an intermittent zero at 1:12... I encounter similar ones

xet303, 2025-03-20:
Buy yourself a $5,000 frequency counter and you'll see these kinds of things happen. What's already been done for below 1000 is already magnificent.
Please, GitHub is not Facebook!

3Dholic, 2025-03-28:
> [AES_V1.7_RC1.zip](https://github.com/user-attachments/files/19360683/AES_V1.7_RC1.zip)
> 
> it is a Bug. please test the new firmware

Today I gave it a try... Another pain in the a... process...

After firmware update the whole telemetry disappeared and also kill switch stopped working... I had to remove all the plugs and get the AES reconfigured again from the radio... S.Port didnt work as before, I dont know why... Whatever...

Results are satisfactory... Now the rpm is way more stable...

3Dholic, 2025-03-29:
Second day of trials... AES stopped killing the engine... Thanks god I was at a very safe spot and skipped the runway exit and killed the engine by the trim 😖😖😖😖😖

I dont know what is going on... Really getting pissed off...

Tried to reconfigure everything from S.port... Everything seems to be working as before, leds are turning from red to green and vise versa... But engine just keeps running...

3Dholic, 2025-03-29:
[https://youtube.com/shorts/sAwrgjjfNHk?si=wvFUjyzI0BR-HAnk](url)

3Dholic, 2025-03-29:
https://youtube.com/shorts/sAwrgjjfNHk?si=wvFUjyzI0BR-HAnk

xet303, 2025-05-27:
Sorry for the delay,
I tried the official version 1.7 this morning.
Everything is working as it should. No more zeros below 1000 or anywhere else!
Problem solved. Thank you very much.