FrSkyRC/RF_Feedback_Group #132: FBUS telemetry sensor limitation TD-R6 + TD-MX

Satcomix, 2025-03-19:
I have a FBUS telemetry problem with two Receivers and did several FBUS tests with my XE, X20S and X20Pro transmitters and receivers. 
With S.PORT there are no problems. I have only FBUS telemetry limitation on two Receivers.

Tested Receivers with S.Port and FBUS protocol:
ArcherPlus R12+ FW: 3.0.1 S.Port+FBUS ok 
ArcherPlus R10+ FW: 3.0.1 S.Port+FBUS ok
ArcherPlus R6 FW: 3.0.1 S.Port+FBUS ok
Archer R6mini FW: 3.0.1 S.Port+FBUS ok
TD-R18 S.Port+FBUS FW: 3.0.2 ok
TD-R10 S.Port+FBUS FW: 3.0.1 ok
TD-R6 FW: 3.0.1 S.Port ok, FBUS limitation
TD-MX FW: 3.0.1 S.Port ok, FBUS Limitation

Tandem XE: 1.6.1 FW 3.0.1 EU/LBT
X20S: 1.6.2 FW 3.0.1 EU/LBT
X20Pro. 1.6.2 FW 3.0.2 EU/LBT

For a few years now, I've been using several oXs_on_RP2040 telemetry sensors, based on a Pimoroni Servo2040 boards, to check my Equipment, for testing and to build telemetry-sensors. A BeiTian BE250 GPS based on an M10 is connected via UART, and a GY-86 gyro and variometer, including an MPU6050, HMC5883L, and MS5611, an MS4525 digital airspeed sensor, and two ADS1115 16-bit ADC voltage modules are connected via the I2C bus.
Furthermore, a clock generator for RPM measurement and a 16-channel SBUS decoder are connected.
With these modules I can normally display up to 45 telemetry sensors on my FrSky transmitters without any problems.
But with the introduction of ETHOS versions 1.6.0, 1.6.1, 1.6.2 and the corresponding RF firmware versions 3.0.0, 3.0.1 and 3.0.2 the problems with the telemetry started.
The FBUS connections of the two Receivers (TD-R6/MX) only start working if only 15-20 sensors are connected to the board and no duplicate IDs are assigned, otherwise no FBUS sensor will be visible.
If more telemetry sensors are available on FBUS TD-R6 and TD-MX, only those from the receiver are displayed in radio.
With all other FrSky receivers I was able to receive up to 45 sensors without any problems.

At first, I suspected the Temp.1 and Temp.2 sensors (Limit 110°C, normally 1100°C), like Udo (strgaltdel), but they're not the culprit.
Secondly, I suspected the data from the GPS and Vario/Gyro, but these modules also work perfectly either individually or together.
I bought two new receivers from Andreas Engel., the Archer Plus R6 and Archer Plus R10+. These work perfectly with older RF firmware and now also with the latest 3.0.1. firmware and the S.PORT and FBUS with 45 Telemetry-Sensors.
Another thing worth mentioning is that the SBUS 16CH portion of the FBUS works flawlessly. All 16CH are displayed correctly all the time.
best regards,
Torsten

KAID-YANG, 2025-03-21:
Hello, we couldn't find this brand in the list of authorized protocols. Could you let us know which manufacturer this product belongs to? As a result, we don't have any samples on hand for testing.

Satcomix, 2025-03-21:
> Hello, we couldn't find this brand in the list of authorized protocols. Could you let us know which manufacturer this product belongs to? As a result, we don't have any samples on hand for testing.

Hello Kaid Yang,
Please excuse me for not mentioning that this is a DIY telemetry sensor project. I thought it was common knowledge. In my opinion, it's irrelevant, since it's worked for the past few years with my RP2040 boards and the TD-R6 and TD-MX receivers. So, something must have changed recently in the firmware of both receivers, as all the other six different receivers I normally use continue to have no problems.
I'll close this issue for the second time and continue researching the problem, as it's somehow related to the number of IDs or the relationship between two IDs or the unit of measurement. It must be something in the S.Port portion/frame of the FBUS.
To do this, I'll have to test and compare all the sensors individually, which will take time.
Thank you very much for your work and effort.
Best regards,
Torsten

closed 2025-03-21 by Satcomix

KAID-YANG, 2025-03-21:
If you have any further information or need more assistance, please feel free to contact us.

Satcomix, 2025-03-21:
> If you have any further information or need more assistance, please feel free to contact us.

Hello KAID-YANG,
I have one more question: Has anything changed regarding the PhyID or AppID or the units of measurement for the telemetry sensors or the Min/Max settings in the system from ETHOS 1.6.0 to 1.6.2 and in RF 3.0.0 to 3.0.2?
Is there a precise list with PhyID, AppID, and transmitted units of measurement for the telemetry sensors in the S.Port protocol?
I've put a lot of work into creating my own S.Port/FBUS list and would like to compare it. It doesn't need to be publicly accessible here. You can also communicate or exchange data via PM.
Best regards,
Torsten