FrSkyRC/RF_Feedback_Group #133: 900 flexible 1W power issue

Burgerman-X10express, 2025-03-19:
I posted this in the RF feedback group. But I now think its an ethos issue.

So heres a copy.

Tx X20 Pro AW
Rx tested R9 mini, R9 SX, etc. Access.
Firmware FLEX latest versions of everything on TX (1.6.2 ethos, 3.0.1 + 3.0.2) and newest flex versions on all 900 receivers.

Issue...
If I set telemetry to show 900 rf Power, and RSSI, VFR and then turn on the Tx, it correctly starts out at 1000mW as it cant yet see a receiver. Correctly.
If I then connect a Rx to the battery (power it up) it then rapidly sees the reciever, which goes green, references the RSSI telemetry etc and drops 900 RF power to 10mW exactly as expected. All good!

Now...
If I disconnect the Rx battery, so the reciever disappears it says "telemetry lost" and RSSI and VFR turn red on the screen. Instead of showing 0, it is displaying the last known telemetry RSSI and VFR. In this case on my bench 99 to 100. And it should now instantly jump up to 1000mW RF output to recover any lost link. But it just stays on 10mW indefinitely. So a suddenly lost link wouldnt be recovered.

Only re-starting the Tx will make it revert to the 1000mW and then it drops to 10mW when receiver is seen.

So that same scenario in the field, maybe "telemetry lost" as you go behind some metalwork, or lost due to RF interference etc, means it will not revert to higher power levels as needed to recover the link. It DOES do that if the signal strength gradually drops as you fly away with range. But not if its suddenly lost due to interference or obstacles!

Which means you may lose an aircraft if telemetry is lost rapidly. This is incorrect behaviour. Maybe because it still "sees" the now frozen (red/lost) high VFR/RSSI values as valid?

So it works as long as RSSI drops slowly. But relies on receiving some low RSSI data to tell it to increase power. If this drops fast enough that no telemetry gets back it stays on low power because telemetry is still incorrectly displayed as previous high values!

Thanks!

P.S. I think you will see the same on a TD receiver, and on R9 FCC not just FLEX as long as you use the 1W (variable power) setting. Untested as of yet.  But I seem to remember this a while ago on FCC mode.

Easy to test, watch and see if telemetry jumps back to 1W as it should when a Rx is switched off.

Burgerman-X10express, 2025-03-20:
Also tested with rf feild strength meter in case its just the displayed value that doesent update.  And its not. Stays low.

KAID-YANG, 2025-03-24:
Hello, the reported issue has been forwarded, and the new version will include a fix for this problem.

SmartTommy, 2025-04-29:
TD-ISRMPro_3.0.3 firmware fixed this, please update the this.

[TD-ISRMPro_3.0.3.zip](https://github.com/user-attachments/files/19954521/TD-ISRMPro_3.0.3.zip)

Burgerman-X10express, 2025-04-30:
Is this for the X20 PRO AW ??