FrSkyRC/RF_Feedback_Group #135: Ethos 1.6.2 - Random "sensor conflict" message

Didier-L, 2025-03-27:
Since 1.6.2 (not sure 1.6.1), I get random "sensor conflict" message during a flight. I get only one message during the flight.
Nothing changed since 1.6.0 on the aircraft side (rx, servos, sensors, ...)
I get these messages with rx using or not using Frsky servos connected in FBus, so it is not linked to it.
I have at least 3 different models with the same issue (not tested the others, particularly the TW ones)...)
Tx : X20 PRO Ethos 1.6.2 EU   / RF module fw 3.02
Rx : Archer Plus SR10 fw 3.01

List of the sensors on one model, esc running BlHeli sending telemetry via S.Port
RX                18 F101
RSSI             18 F101
VFR             1A F010
Tension Rx   18 F104
ADC2          1A F103
AccX            1A 0700
AccY            1A 0710
AccZ            1A 0720
Roulis          1A 0730
Tangage      1A 0730
Tension ESC     06 0B50
Intensité ESC   06 0B50
Conso. ESC      06 0B60
ESC RPM         06 0B60
Temp. ESC       06 0B70

A friend of mine has the same issue with a X20RS, Archer Plus SR12 and FAS150 connected in FBus.

Didier

craigywsm, 2025-03-27:
> It will be shared in the Discord room after the first smoke tests

Sorry is this an RF / Ethos issue?

bsongis, 2025-03-27:
RF (internal module)

SmartTommy, 2025-05-06:
The TD-ISRMPro/RS 3.0.3 fixed this issue

yak-54, 2025-05-06:
Still getting Random "sensor conflict" message on  my x20 pro using Archer s6r with Mlvss