FrSkyRC/RF_Feedback_Group #136: td sr12 issues no stabilization no rxbatt voltage

williajgr, 2025-04-03:
no stabilization   td sr12 receiver. only auto level
 no telemetry for rxbatt1 or 2 or current. all other telemetry seems to be there
calibrated all sticks

x20 rs transmitter 1.6.2

tdsr12 receiver script 3.0.6 remote version 3.0.1

wonder if the receiver is defective.

KAID-YANG, 2025-04-09:
Hello, the TD SR12 only has one RXBatt and does not have stabilization functionality.  
Have you set the Gain on CH13?

williajgr, 2025-04-10:
I don't have any rxbatt

KAID-YANG, 2025-04-14:
Please try reflashing the receiver using an S.Port cable. We have tested it and were not able to reproduce the issue with missing RXBatt. More information is needed to investigate further.

KAID-YANG, 2025-04-14:
Please provide the information from the INFO screen.

williajgr, 2025-04-16:
> None of my sr receivers will update by cable.Sent from my iPhone

Ok solved the problem with it not having stability.  When I did the last update on my transmitter, it changed pot2 to pot1, so I had  it set to 0.   I am thinking about never updating the firmware.  It has cost me big problems in the past.Sent from my iPhone

closed 2025-04-27 by SmartTommy