FrSkyRC/RF_Feedback_Group #137: Rf007    TW

stubugs, 2025-04-06:
X20rs 1.6.2 int module  3.0.2
Rf 007 3.0.1
Rf007 registers and binds great
Flash twr6 with  3.0.2 and will not register or bind

rburrow87, 2025-04-06:
Make sure you flash it with the TW R6FB firmware, not regular TW R6.

stubugs, 2025-04-07:
I tried again down loading the file twr6fb 3.0.2 put it on x20rs firmware file tried flashing external device 
Would not bind still
So I tried ethos suite plugged rx into tx and flashed with download center and flash not down load and it worked straight away registered and binded with 3.0.2  twr6fb
So I'm not sure if this is a problem  now or not

KAID-YANG, 2025-04-09:
Hello, the RF001 TW version has a built-in TW R6 receiver. Please try upgrading the receiver using an upgrade cable.

closed 2025-04-09 by stubugs