FrSkyRC/RF_Feedback_Group #138: Telemetry - no voltage reading from TDSR12 receiver on X20pro

sigurdma, 2025-04-07:
After update TDSR12 to 3.01 it does not have Volt in telemetry. All other telemetry is present.
I have reset to factory two times ans also reinstalled 3.01, unregister an unbind many times.
Still no voltage in telemetry. I have X20pr on 3.02
A buddy has no problem reading dual vaoltage from XT30 inputs on Rx. he has X20S on 3.01.

KAID-YANG, 2025-04-09:
Are you referring to the voltage data from ADC2?

sigurdma, 2025-04-09:
No there is no rxbat in telemetry from TDSR12. I see that more has the same problem, and also with X20S.
So I conclude that it is the firmware in receiver that has a bug.

wfield0455, 2025-04-12:
Same problem. X20S Transmitter 1.6.2 / 3.0.1 firmware. TDSR12 running 3.0.1 firmware. Multiple binds, factory resets, etc.  Deleted and rediscovered telemetry and there is no RxBat sensor detected.

KAID-YANG, 2025-04-15:
We are using the X20 with ETHOS version 1.6.2, RF module version 3.0.1, and TDSR12 firmware version 3.0.1. After testing, we were unable to reproduce the reported issue. Could you please provide the hardware version number for further investigation? [image]

wfield0455, 2025-04-15:
As I mentioned before, I have downgraded everything to 1.5.19 but in case this is helpful I have included the current software / hardware versions of my x20s and tdsr12.V1.5.19TD-ISRM HW: 1.4.0 FW: 2.2.7(FCC)TD SR12HW: 1.1.0 FW: 1.0.11Hope this helps you figure out what is going on..WayneSent from my iPhone

wfield0455, 2025-04-16:
I have an additional update. Since you mentioned you were not able to reproduce the no "RxBat" sensor issue using an X20 I decided to upgrade my TDSR12 receiver to 3.0.1 again and test it against my X20 Pro which was still running V1.6.2 / TDISRM-Pro 3.0.2. I upgraded the receiver using the Ethos Suite "FLASH" option, performed register, bind, factory reset, register bind, delete telemetry, rediscover telemetry, exactly as I had done before and this time the RxBat sensor was discovered and worked properly. I had tested this before using my X20Pro following the exact same procedure and it did not work before. The only difference I can see is that I previously upgraded the receiver using internal OTA on my X20S using the file I downloaded from Ethos Suite and unzipped to my X20S.

At this point I decided to upgrade my X20S to V1.6.2 ISRM 3.0.1, registered and bound the receiver, deleted telemetry and rediscovered it and the RxBat sensor is now displayed on my X20S as well and everything is working as expected.

The only thing I did differently this time is FLASH the receiver using S.Port cable from my X20Pro using Ethos suite rather than internal OTA from my X20S. Perhaps the file I used on my X20S is corrupted ? I don't know but now everything is working properly. Hope this information is of some use.

sigurdma, 2025-04-16:
That was positive news. I decided to do the same, use Ethos to flash (not download the file). No it works fine here too.
At the same time I think I discovered the reason. The download file is 3.01 RC2 and the flash file is 3.01 (release).
I am sure that is the reason, there is no other explanation.

wfield0455, 2025-04-16:
That’s good news! I’ glad the info I provided was useful.Sent from my iPhone

KAID-YANG, 2025-04-21:
> That was positive news. I decided to do the same, use Ethos to flash (not download the file). No it works fine here too. At the same time I think I discovered the reason. The download file is 3.01 RC2 and the flash file is 3.01 (release). I am sure that is the reason, there is no other explanation.

This is just a file naming error on our part. It is actually the official version that has been tested. If you're experiencing issues with OTA updates for the receiver, we recommend using an S.Port cable to connect the receiver to the transmitter and reflash the firmware manually.

sigurdma, 2025-04-21:
Of course I tried to use S.port on X20pro too with same result, no Rxbatt. So at last I tried (as wfield455 did) flashing direct via Ethos and then it was ok. 
I have a friend with same problem, so when I meet him I will try again with S.port before flashing via Ethos.

wfield0455, 2025-04-21:
I had also tried flashing via cable using the file I had unzipped to my transmitter. I tried multiple times from both my X20S and X20 Pro and got no RxBat telemetry. It's only after flashing directly from Ethos Suite that is started working properly.

ltdorn, 2025-04-26:
X20pro AW: i also was having telemetry issues in general with the 3.01 rx updates, fw 1.62 and tx latest tx module update..  all my telemetry values on  my turbine jet were just rapidly flashing like it was polling telemetry request to fast on the telemetry screen... one the display my voltages / VFR were all just flashing intermittently on the screen.. the countdown timers looked normal.... I also have a X20 Pro that had NOT been upgraded to anything above 1.6 and the telemetry looked fine when bound to the same rxs that was giving my X10Pro AW fits.. I also did what others did to try and trouble shoot the x20pro AW.. including downgrading fW to 1.6, and the swiping all LUA scripts, then finally completely swiping my TX to factory with no models or LUA scripts installed with FW 1.6 then i bound a TDSR12 and TDSR18 by themself.. and the telemetry displayed normal.  then i slowly upgraded the 20AW to the current firmware's with no lua scripts or old models reloaded...  long story short... if i restored my old saved 1.62 models back to the TX.. the telemetry on both a TDSR12 AND TDSR18 went wonky...  if i created a brand new model on 1.62 then the telemetry looked fine on my RXs.... so from what i could tell is the old models that have been upgraded from 1.6 to 1.62 are causing the telemetry to act wonky....  I really dont want have to rebuild all my 20 plus models on my 20AW to get telemetry to work..  already ran down my two RX battery trying to trouble shoot this issue for the last 5 hours... so they are on the charger and i am burned out for the night... i might try that flash both of those RXs with the ethos suits vice OTA and see if that make a difference...  but anytime i try to use any restored version of any of my models telemetry goes crazy unless i build a brand new model in 1.62 and the telemetry looks fine.

ltdorn, 2025-04-26:
> X20pro AW: i also was having telemetry issues in general with the 3.01 rx updates, fw 1.62 and tx latest tx module update.. all my telemetry values on my turbine jet were just rapidly flashing like [...]

Note more trouble shooting this morning... including trying to bluetooth transfer a model from my old 20pro to the 20proAW and still get the same telemetry issues of flashing between white and red dashes...   the ONLY way i have found to get telemetry to operate "normal" with 1.62 and tx module 3.02? is to create a brand new model.......  so is there new model template in the different FW versions that can mess up the telemetry processing on the transmitter?  just annoying because i want to see my turbine telemetry because i use it tell me if there was a flame out and the turbine temp start to rapidly decrease in the air... can her the turbine cut out in the air with others flying gassers or other turbines...

sigurdma, 2025-04-26:
If its only telemetry that is problem, have you tried deleted telemetry and scan again? Could also reset rx.