FrSkyRC/RF_Feedback_Group #142: no 3.0.3 RF Firmware for TD-ISRM (Tandem x18s / x20s) ?

strgaltdel, 2025-05-01:
Hi,

since today's official newsletter many people will realize that 3.0.3, which contains solution for issues,
#135
#133

 is not available for their tx
will there be a 3.0.3 for td-irsm ?
(in case these issues did not affect td-isrm, a hint maybe usefull, so it could be explained in forums)

regards

Alex00101, 2025-05-01:
Checked X20RS, is is available from FrSky website download section, probably not in the Suite at the moment

Dsrenger, 2025-05-01:
He is asking about the TD-ISRM module. 

That's not the X20RS

applewood, 2025-05-04:
Same, I can’t see 3.0.3 TD-ISRM update for either my X20 or X18

SmartTommy, 2025-05-06:
TD-ISRM   3.0.3  will be released asap

applewood, 2025-05-06:
> TD-ISRM 3.0.3 will be released asap

Thank you.  Will you post here when it’s available?

applewood, 2025-05-11:
Hi, any news on when this firmware will be released?

ChrisKhosravi, 2025-05-25:
Looks like this hasn't been released yet. I wonder why.
I also have sensor conflict issues that I can't seem to figure out.

applewood, 2025-06-11:
Any update please?  It’s now 6 weeks?