FrSkyRC/RF_Feedback_Group #147: ACCESS Archer R4 not bind

Maco33, 2025-05-26:
The Archer R4 receiver worked normally in the model. After changing the FW several times. via OTA, when everything went normally, it can no longer be bind. Registration goes normally, but binding cannot be done. The procedure was verified with another receiver and everything worked. Tried different FW. installed via S.Port, different FW. TD-ISRM, but still nothing :| Any ideas or recommendations? (FW.1.6.0, 3.0.1, 2.1.14)

mawzthefinn, 2025-06-04:
Reflash wired, that is a symptom of a bad OTA flash/

Maco33, 2025-06-04:
Yes, I tried several times with different fw. via wire, but still couldn't make a bind. In the end it helped that I steamed another receiver into the transmitter in place of the non-working receiver and tried the damaged one blind to the new aircraft. But the OTA is probably damaged, the flash fw didn't work again. I'll have to investigate.