FrSkyRC/RF_Feedback_Group #150: TD-SR18 X20Pro AW 1.6.2 ISRM 3.03 Calibration Fail

Joetrin, 2025-06-01:
Radio is X20Pro AW in 1.6.2 on isrm 3.0.3
Stabilization 3.06

TDSR18 3.0.3

6 Axis calibration fails consistently on the 3rd step.
Works on rx right side up and level
Works on inverted
Fails on label facing left

When it fails, I get "telemetry lost" message and the calibration screen hangs on "Wait until calibration comoletes" dialog box.

When we back out of that screen to the main stabilize screen, it shows no connection to the rx.  I have to cycle the rx power in and off in order to get the radio to recognize the rx again.

Its repeatable.

Joetrin, 2025-06-04:
I will unplug everything and try again.  Ill do a factory reset this time.

Ill post results here.

Thank you

we checked the firmware and it works

https://github.com/user-attachments/assets/dc2b4f5e-0057-48ba-a247-ad4ef709bbd7

SmartTommy, 2025-06-19:
> I got it to work by NOT following the calibration instructions. The TDSR18 does not have pins in the front, and the orientation images and wording are not describing the actual need position of the RX.
> 
> Using the Antennae as reference for pins, I was able to calibrate the RX.. I believe the new firmware has an expected Orientation therefore if the orientation does not match the expected outcome, it fails. Perhaps a warning message will help too, describing the expected orientation of the RX.
> 
> Thank you for your help.

We will add more instructions and descriptions to this feature for the convenience of users

closed 2025-06-19 by SmartTommy