FrSkyRC/RF_Feedback_Group #152: RB25S can not be configured

wfield0455, 2025-06-04:
Following upgrade of RB25S to RB25S_20240511 and TDMX receiver to V3.0.3 firmware I can no longer configure the RB25S. Menus are all
greyed out and in basic config menu it shows IMU off but since it's greyed out it can't be set to "Basic".

I upgraded the TDMX to V3.0.3 using Ethos Suite. PErformed performed factory reset and performed register and bind operations. I then set options to FBUS for communications with the RB25S. I then updated RB25S to BB25S_2024_0511 firmware using Ethos Suite and the upgrade completed without issues. I also copied the latest RB25S LUA 3.0.0 files to the scripts directory of my X20S. I Tested servo operation and servos operate properly. I  then used Device Config-> Flight Safety -> RB25S and all selections show all menus are greyed out and nothing can be changed.

X20S V1.6.2, latest module code (TDISRM 3.0.3 FCC).
TDMX receiver firmware V3.0.3 connected to RX1 of RB25S set to FBUS
RB25S firmware RB25S_20240511.

sigurdma, 2025-06-06:
I downloaded the 3.03 for TDMX. Readme file says: This Package is for receiver TD SR12 receiver's firmware update.
Try the 3.01 and se if you get the same fault.

wfield0455, 2025-06-07:
Thanks for the suggestion but  tried 3.0.1 still and it still doesn’t work. Someone suggested trying to configure the RB25S via x20 s. Port connection and that doesn’t work either. Not sure if this is a bug or the firmware update for the RB25S failed and backed my RB25S or what but I can’t communicate with the RB25S any longer by any means. I may try borrowing a friends x20s running 1.5.19 and see if I can recover it using old firmware and lua scripts Sent from my iPhone

wfield0455, 2025-07-22:
After  updating RB25S firmware to version 3.0.5 RB25S 2027-07-17 I can now configure the RB25S properly again.

closed 2025-07-22 by wfield0455