FrSkyRC/RF_Feedback_Group #154: AES II T6-T7 telemetry issue

3Dholic, 2025-06-06:
https://www.facebook.com/share/p/1FhABWjjX6/?

Posted on facebook group...

Trying to configure AES II through FBus however, when I discover the telemetry (after deleting all), it only discovers the sensors from T1 to T5 only, however T6-T7 doesnt appear...

Since T1 to T4 are dedicated to PT100...

T5 to T7 are dedicated to TEMS...

How to get telemetry through T6 and T7 ?

Note: It keeps calling "Sensor Conflict"

SmartTommy, 2025-06-06:
could you tell us your ethos , Tx  and Rx software version?

3Dholic, 2025-06-06:
1.6.2 ETHOS

3.0.1 TDR 18

SmartTommy, 2025-06-06:
The next ethos version  will support AES_V2.   
"Sensor Conflict" be fixed,please update the "TD-ISRMPro" or "TD-ISRMRs" to 3.0.3

3Dholic, 2025-06-06:
So, updating only the ISRM will resolve the problems without updating any of the receivers, right ?

SmartTommy, 2025-06-06:
> So, updating only the ISRM will resolve the problems without updating any of the receivers, right ?

yes, the ISRM 3.0.3 firmware fixed the "Sensor Conflict"problem.

3Dholic, 2025-06-07:
Today, tried the AES with the updated Module Firmware (see the video in the link)

I dont understand why but T6 and T7 still doesnt appear...

Note: We plugged the RPM later on and there no problem with it, so dismiss the missing RPM telemetry in the video. However T6 and T7 is still missing...

[https://youtu.be/-TLFxqccj0s?si=Zbflqu0ZHZbDa42S](url)

SmartTommy, 2025-06-19:
> Would there be any feedback from the factory team regarding the issue ?
> 
> Still not able to resolve the problem

ETHOS 1.6.3 will add the new AES sensors, will be released soon

3Dholic, 2025-06-19:
> > Would there be any feedback from the factory team regarding the issue ?
> > Still not able to resolve the problem
> 
> ETHOS 1.6.3 will add the new AES sensors, will be released soon

Very good news... When should we expect the release ? In a few days, weeks or a month ?

3Dholic, 2025-07-03:
**Update 3rd July 2025:**

After updating the device with the newly released firmware version 1.6.3, T6 and T7 are now being detected. The issue appears to be resolved. I just wish this firmware had been released prior to the launch of AES II...