FrSkyRC/RF_Feedback_Group #161: s.port ESC telemetry stopped working w/ TW SR12 w/ FW 3.0.3, works w/ 3.0.1

bob01, 2025-06-24:
Scorpion Tribunus III ESC (v16) s.port telemetry no longer working with the TW SR12 w/ FW 3.0.3.
Worked previously w/ FW 3.0.1, works again if rx is downgraded to 3.0.1

- working fine with TD SR10 and TD SR18 w/ FW 3.0.3
- was working fine in same installation w/ FW 3.0.1
- ESC sensors not discovered
- previously discovered ESC sensors not updated

Tx is X20RS w/ ETHOS v1.6.2 + RF FW 3.0.2 
- updated Tx RF module to 3.0.3 - same result, no ESC telemetry from the TWSR12

SmartTommy, 2025-06-25:
[TWSR10_3.0.3_20250625.zip](https://github.com/user-attachments/files/20895821/TWSR10_3.0.3_20250625.zip)
[TWSR12_3.0.3_20250625.zip](https://github.com/user-attachments/files/20895822/TWSR12_3.0.3_20250625.zip)

i add a CRC in the TWSR10/12  3.0.3 firmware, it seems CRC aways failed when use connect to the Tribunus III ESC. now i remove the CRC(3.0.3_20250625), please try.

bob01, 2025-06-25:
Hi Tommy - tested w/ the TWSR12 - it works again!
Thank you very much.
Should we wait for a 3.0.4 with this fix?

[image]

mwilson007, 2025-06-25:
I also tested on a TWSR12 and it works fine for me as well.