FrSkyRC/RF_Feedback_Group #174: Poorer RSSI and VFR values after update to RF 3.0.X

EngelMT, 2025-08-26:
I have flashed my X20 Pro and my TDR-10 as well as the TDMX to the latest version. I fly my jets with 2 x TDR-10 and 2 x TDMX. I have also set "Individual RSSI alert per band".

When I received an RSSI alert for 868 MHz during a flight with my jet at a distance of about 150 m, I looked at the log files after landing. The result was not encouraging. The RSSI and VFR values for 2.4 GHz were always better than those for 868 MHz. Even though I was flying with 500 mW output power. I then switched off "Individual RSSI alert per band". Then I no longer received the warnings. Overall, I can see a big difference when comparing the log files for ETHOS 1.5.19 with RF 1.1.13.

1. the RSSI and VFR values are worse overall at comparable distances.
2. the RSSI and VFR values are significantly worse than the values for 2.4 GHz despite 500 mW transmission power in the 868 MHz band.

FrSky has received the log files.

For me, this is not yet the final solution. I have flashed back to 1.5.19 and RF 1.1.13.

Since there are also some reports about this in our German FrSky forum (9000 members) and there seems to be the same problem with the TW protocol, it is certainly not an isolated case for me. An acquaintance also complains about it. He flies large gliders and keeps getting warnings.

I hope FrSky can reproduce the problem and find a solution.

EngelMT, 2025-08-28:
Hi Simon,

I think FrSky will working on it because they told me to open the issue to maybe get more information.

I will do a real range test later with 1.5.19 and the last RF and receiver firmware before 3.0.0. I'm curious to see if I still have bad values as you describe. I'll get back to you.

By the way, we are also a Service Centre Europe. We are also happy to help Norwegians. For us, Europe does not mean the EU, but a country in Europe. If you are really considering upgrading the Pro to ProAW, just send an email to service@engelmt.de and we will look into the costs.

EngelMT, 2025-09-01:
On last Thursday I do a real range test (not in range test mode) and 1.5.19 / 1.1.13. And on the weekend I fly my Mirage on an Airshow. All works fine now. The range test is working up to 700 m with 2.4 GHz without problems. When I had a hill between me and the model, 2.4 GHz was of course gone. But 868 MHz always worked. Even when I was a good 100 metres lower than the model behind the hill.

The following screenshots of the RSSI and VFR values show that 868 MHz is clearly better. That's the way it should be.

[image]
[image]

With ETHOS 1.6.3 and RF 3.0.3, 868 MHz was always worse than 2.4 GHz and there were always RSSI warnings on 868 MHz.

I would like to repeat once again that I never had a control failure with 1.6.3 / 3.0.3 either. Maybe the display of the values is simply wrong?

Just for your information. Today we had a customer's X20 Pro transmitter with an Archer Plus receiver operating in ACCESS mode here in the service centre. The customer complained about the same problem and we unfortunately had to confirm this. During the tests, we also noticed that the values fluctuated greatly. We then flashed this transmitter back to 1.5.19 and 1.1.13 and, lo and behold, in tests in range mode the values at a distance of 100 m are almost twice as good as with 1.6.3 and 3.0.3.

I would therefore like to ask FrSky once again to take a very close look at the problem. As I said, I still suspect that the transmission is OK, but that the transmission of the telemetry data has a problem.

EngelMT, 2025-09-09:
@Shang2017 any news about this issue?

EngelMT, 2025-09-26:
I realised that I did not mention above that we use EU LBT here. Please test with it as well. Because I don't hear any problems from my friends in America and Israel. But they also use FCC.

PKhair, 2025-09-29:
I noticed similar issue with 900Mhz band after upgrading from Ethos 1.5.x to 1.6.3 along with the update of ISRM and RX to FW 3.0.3. Prior upgrade, the 900Mhz band was always the strongest in RSSI and VFR, after upgrade 900Mhz looks very weak, always below the 2.4GHz band (it was the opposite prior upgrade). While performing range testing, 900Mhz band drops very quickly and telemetry stops working while 2.4Ghz stays relatively strong which is really an unexpected behavior. The only difference from what is reported earlier in this thread is that I noticed 900MHz VFR drops significantly along with RSSI, whereas EngelMT only reports RSSI issues.

After upgrade to 3.0.3 I have also noticed systematic failure while performing OTA update of TD SR-18 receivers. OTA upgrade fails randomly during flashing anywhere between 30% and 80%. Only way to recover is wired flashing of receiver using transmitter S-Port. I have tried OTA upgrade of 2.4GHz access receivers (SR8 Pro) and it still works fine.

I have been running into these issues with 2 different large jet models which used to work perfectly fine prior update. Both are equipped with TD-SR18. I am really not comfortable large and expensive planes before understanding what is going on with RF link and FW 3.0.x

Detailled information:
- X20 Pro: Ethos 1.6.3, TD ISRM Pro FW 3.0.3
- Tandem SR-18: FW 3.0.3
- EU LBT mode, 25mW output power on 900Mhz band

PKhair, 2025-10-07:
@SmartTommy any outcome from your code review? Could you please provide an update? This really seems like a major issue for many EU customers. Thanks !

EngelMT, 2025-10-20:
@SmartTommy  Some customers out of our forum asked me about an update to solve the issue too. So please check the issue again?

By the way. Can all those who have written here that they also have the problem write exactly with which radios and receivers the problem occurs. Perhaps it is a specific piece of hardware that is causing problems. E.g. the X20 Pro.

PKhair, 2025-10-20:
X20 Pro: Ethos 1.6.3, TD ISRM Pro FW 3.0.3
Tandem SR-18: FW 3.0.3
EU LBT mode, 25mW output power on 900Mhz band

Similar behavior on 2 planes running the same setup.

EngelMT, 2025-10-30:
@SmartTommy Any news to this issue? Customers are uncertain and FrSky should find a solution! I already suspected that it could depend on the RF hardware. Just as @Dsrenger suspects.

However, I must say that we have now received feedback from X18 transmitters with TW or ACCESS receivers in our forum, which mention a shorter range with the current firmware and ETHOS 1.6.3. Could it be that there is no real range problem here, but rather an incorrect calculation of the RSSI and VFR values, which in turn leads to poor log files and early warnings?

Florentus, 2025-11-12:
I’m starting to think we’ll never get a response from FRSKy to this issue. No doubt that they are very busy, but they should however take into account a little more the problems that their clients submit to them especially when several people have reported this issue.
What else to do?

Florentus, 2025-11-14:
@EngelMT I agree with you, all manufacturers are facing problems, which is unfortunately quite common. FrSky has many assets, including the ETHOS software which is undoubtedly one of the best in RC world (at least for me). I have never lost a single plane because of these radios. What is regrettable is the behavior of the after-sales service which, despite all the feedback from customers, especially yours who manage the after-sales service for Europe, gives the impression of ignoring the problems. It is undeniable that the problem you have noticed is not due to misuse on your part. I think they could use an X20 PRO in LBT mode to perform the tests and finally remove these doubts that persist since this new RF version.  I use a 20 PRO AW which may not be concerned, but as a precaution, I don’t dare to update it to 1.6, it’s really a shame.
In any case, thank you for your help

EngelMT, 2025-12-02:
@WhiskeyGolf

I can reassure you that there is a test firmware available. I have already flown with it. The log files look much better. With very few exceptions, the RSSI value of 868 MHz was always better than 2.4 GHz. And the VFR value was almost always 100 per cent. Just as you would expect. However, the flights were only at 25 mW power. I want to do a few more flights at 200 mW. That's the transmission power I normally use for my large jets.

EngelMT, 2025-12-02:
@T16-Pilot 

It's not certain that this only affects the X20 Pro. It could just as easily affect the X20 Pro AW. Unfortunately, there are conflicting reports on this. And does the upgrade to AW Pro really include a new RF board? I wouldn't be so sure. To me, there's really no reason why FrSky should pay for the upgrade.

I'm still using an X20 Pro myself and would have preferred it hadn't been replaced by the X20 Pro AW so quickly. But the advantages of the AW are limited, so I'm not really upset.

You're right, FrSky doesn't communicate these kinds of issues the way we'd expect. But there's certainly a big difference in thinking between Europe and China.

T16-Pilot, 2025-12-02:
@EngelMT I know that the RF board is being changed when "upgrading" the X20 Pro to AW.

I saw a email from Frsky Ben with the list of components being swapped when upgrading a X 20 Pro to AW. It was a long list... and no, sorry i don´t have that list. A friend of mine email Frsky Ben and got a complete list with all the components that where changed.

Yes, Frsky should pay to fix a product they sold. Here in Norway there is a 2 year warranty witch can be extended to 5 years if a product is men´t to live at least 5 years. So yes, Frsky need to fix this issue.

I think that Frsky know that there is a problem with the hardware in X20 Pro because you don´t change the RF section on a brand new top of the line radio if every thing is ok? 

How long was the X 20 Pro on the market? 6 months ish..? before the AW came and they cancel the X 20 Pro. 

X 20 and X 20s share the same RF section of the "first generation" ACCESS.

X 20 Pro has a custom "one of" RF section.

X 20 R/RS/AW all share the same brand new RF section.

There must have been a reason for why they cancel´d the X 20 Pro so quickly?

Regards
Simon

RealTadango, 2025-12-05:
@EngelMT Did you also test with Ethos 1.6.3 and RF 1.1.13? That is what i use at the moment. Just to eliminate Ethos because you keep changing 2 firmwares.

WhiskeyGolf, 2026-02-06:
FrSky has just officially released version 3.0.4.👍🏻