FrSkyRC/RF_Feedback_Group #175: AES V1 - Temp reading not correct

LIBRAPDB, 2025-08-31:
Hi guys, I'm setting up an AES V1 ( advance engine suite) with a 5 cylinder gas radial.
Everything fine, but as long as I connect the temperature sensors ( all of them) the temperature starts from 88° Celsius, and obviously they are not at 88° but at room temp. If I try to heat them, the reading goes up so they are working, but they are starting high ( see picture attached on TEMP 5)
any ideas? thanks in advance

[image]

xet303, 2025-09-01:
I verify here with 3-4 NTC spares (sensor that Frsky uses) .At 21 degree C the 3 have a resistance varying from 107.2 to 110.4 Kohms.
If your sensors have a similar resistance and you get +88c the internal calibration of your AES is probably outside the fields!
I also get -44C for an unconnected sensor.

pstasek, 2025-09-07:
We went through some troubleshooting with @LIBRAPDB via WhatsApp. I'll fill in a few more details regarding this issue that may narrow down the issue.

Radio used is Kavan V20, Ethos version 1.6.3, receiver is Kavan R8, AES firmware is 1.7.0. 

The value 88° roughly corresponds to the actual temperature in Fahrenheits, despite the units are set to °C. If the unit is set to °F, the value reads 140 as expected.

Tried a Factory Reset in the AES config script.

I recommended using the TW R12 instead of the Kavan R8 receiver as it would suit the model much better to rule out the Kavan receiver firmware.

LIBRAPDB, 2026-02-15:
about the calculated sensors:

I choose the AES Temp 1

then on ''subtract/multiply/divide'' its asking me for a category

which category should I choose in order to add the numbers? ( 32 on subtract , etc)

can you also upload the screenshot of the first values of the calculated sensor?

formula, unit, decimals, interval , etc

thanks

pgaryga, 2026-02-16:
Sorry, I'm not sure what you mean, the calculated sensor is it. You add a value widget and select the calculated sensor as the source.